haplokuon/netDxf #224: Exception. Quotation marks in the font name of text styles

iborzenkov, 2020-10-21:
I have a dxf file in which the font name of text styles is enclosed in quotation marks. Autocad works successfully with this text style. 
However, NetDxf generates an exception when reading this file at method **DxfReader.ReadTextStyle** in the string: 
`ShapeStyle shapeStyle = new ShapeStyle(Path.GetFileNameWithoutExtension(file), file, height, widthFactor, obliqueAngle);`

There is a suggestion to pre-trim the quotation marks in the font file name of text style.
[Link to the DXF file](https://drive.google.com/file/d/103g78ps81I0r0Bj8HIpH7X5wdb7EmYgQ/view?usp=drivesdk)

haplokuon, 2020-10-23:
How do you end up with a Style that points to a file with quotation marks when those characters are not allowed as part of the file name by Windows?

iborzenkov, 2020-10-23:
Unfortunately, I do not know how this font was added to the dxf file. 
I will try to clarify the script how this font name was added. But in the end, DXF is a test file and the font name could be written in a text editor, because this is not prohibited. 
The sad thing is that our users have a DXF file template that they use to create their drawings. There are thousands of these drawings and they cannot be processed in NetDxf.

iborzenkov, 2020-12-14:
(+) Not observed in version 2.4.1 release

iborzenkov, 2021-01-19:
A little late, I studied the solution proposed in 2.4.1 release. It seemed doubtful to me.
The solution was that if incorrect characters were found in the file name, the entire file name was replaced with a certain constant string "FILE_NOT_VALID".
This is really a better solution than throwing an exception. It is suitable for general cases.

But in my original scenario, the file name is correct. It is only framed with quotation marks. This is what the Windows operating system does in some cases. I don't expect "FILE_NOT_VALID" to be used instead of my file name.
I suggest adding a line `file = file.Trim(new[] { '"' });`
```
// basic check if file is a file path
file = file.Trim(new[] { '"' });
Debug.Assert(file.IndexOfAny(Path.GetInvalidPathChars()) == -1, "File path contains invalid characters: " + file);
if (file.IndexOfAny(Path.GetInvalidPathChars()) != -1)
{
    file = FileNotValid;
}
```

haplokuon, 2021-01-24:
As some other characters the quotation marks are not allowed as part of a file name, sometimes they are used by search engines as the characters * and ? but never as part of the name. What I forgot to do is to allow changing the File property of a ShapeStyle.