haplokuon/netDxf #225: Hatch Edge fails on Arc?

deddolo, 2020-10-25:
I have a hatch composed of straight Lines and a portion of an open Arc.

When traversing the edges in the boundary path, they are all type line and the arc is ignored. Am I doing something wrong?
Funny enough, one edge has a random Z coordinate in it.

[image]
[image]

haplokuon, 2020-10-26:
The edge HatchBoundaryPath.Polyline works like a LwPolyline, the Z value represent the bulge at that vertex. The arc edge you are looking for corresponds to that vertex with a non zero value for the Z coordinate.

deddolo, 2020-10-26:
Okay. Thank you for the help. I ended up converting to LwPolyline and calling PolygonalVertex() to get a segmentation.

closed 2020-10-26 by deddolo