haplokuon/netDxf #228: LibreCAD generated files don't load.

computergeek1507, 2020-10-30:
I exported the attached files with the LibraCAD Nightly and neither load. One is AutoCAD2007 and the second is AutoCAD2004 format.

[dxfs.zip](https://github.com/haplokuon/netDxf/files/5462458/dxfs.zip)

haplokuon, 2020-11-01:
Both files Untitled3.dxf and untitled4.dxf are bad files, even AutoCAD will crash trying to load them. The problem lies on writing a non hexadecimal number in code 340. You should report this bug to LibreCAD. The official DXF documentation is awful, but the only thing that makes it clear is the code-value correspondence. I am adding several new checks for invalid values in buggy DXF files, but, in any case, this kind of errors in a DXF file are unacceptable.

computergeek1507, 2021-02-17:
Is the 340 with the "STANDARD" the problem? The second 340 is a value for the 5 code above it, I think.

[image]

[image]

haplokuon, 2021-02-17:
Correct. The data in a DXF is always stored in pairs (code-value) therefore odd line numbers are always codes, while even lines are values. Code 340 must be followed by an hexadecimal number, in your second image it just happens that the handle for the LwPolyline entity, code 5 is 340.