haplokuon/netDxf #229: Problem with the line type after saving

FilatovAV, 2020-10-31:
After saving the dxf file, lines are no longer displayed correctly if they use shape elements. I attach a project in the Studio with a dxf file (2000.dxf), the program adds a line and saves the file under a different name. Note the line 2g_474_2a.
The file was created in AutoCAD 2017.
[DxfTest.zip](https://github.com/haplokuon/netDxf/files/5469644/DxfTest.zip)

FilatovAV, 2020-10-31:
Added a shx file to the project for the line type.
[DxfTest.zip](https://github.com/haplokuon/netDxf/files/5469693/DxfTest.zip)

FilatovAV, 2020-10-31:
After loading the line type, there are problems with the code page, but the line type itself is not displayed correctly.
Updated the project and added the line type description file test.lin.
[DxfTest.zip](https://github.com/haplokuon/netDxf/files/5469763/DxfTest.zip)

[image]

haplokuon, 2020-11-01:
When working with shapes, as an entity or in a complex line type, you need also access to the SHP file. See the sample methods "Shape()" and "ComplexLineType()" in the TestDxfDocument project. The SHX flie is supposed to be a compiled, a binary form of the SHP text file; but I had no luck trying to read the SHX file, there seems to be additional information not present in the SHP file; or perhaps I am just doing something wrong. If there is someone out there that know how to read SHX files let me know.

About the description in the line type, save the LIN file as UTF8, you can use Notepad++ to convert it from the actual ANSI code. Even the AutoCad version I tested that LIN file doesn't like the description in Cyrilic, not in ANSI or UTF8, but this might have something to do with the AutoCad installed language.

closed 2020-11-01 by FilatovAV