haplokuon/netDxf #230: Layer name when object is annotative

FilatovAV, 2020-11-03:
Hello

I came to You with a new problem. =)
If the object is set as annotative, the object layer becomes strange.
For example, if it is on layer "0", then its name will be "0 @ 1". 
I attach a project with a demonstration of the problem.
[DxfTest.zip](https://github.com/haplokuon/netDxf/files/5479443/DxfTest.zip)

haplokuon, 2020-11-04:
Sometimes AutoCad likes to write in the layer names, and perhaps in other places too, additional information. The details for your case I ignore them, but I know of a similar case. When a DWG is added to a another drawing as an external reference, layers of the attached DWG appears as "referenceName|layerName" in the new drawing. When a DXF with a external DWG is loaded, the layers with such names will be skipped since the character '|' is not allowed. In your case the layer names are valid so they are added.

I do not know how it works for your case, but if that layer name "0 @ 1" hides some kind of code, I do not understand why they are taking that route since, since that string do not contain invalid characters and a user may have named a layer with that same name and perhaps interfere with what it is trying to code. No idea.

Not every thing is supported and trying to load such objects may be end up partially or not loaded at all. At the moment I cannot do anything about it. Once again the official DXF documentation is pure crap, it leaves too many thing in the shadows. Why bother to create a drawing exchange format if you do not care to properly document it. And now I am beginning to rant.

haplokuon, 2020-11-04:
Sometimes AutoCad likes to write in the layer names, and perhaps in other places too, additional information. The details for your case I ignore them, but I know of a similar case. When a DWG is added to a another drawing as an external reference, layers of the attached DWG appears as "referenceName|layerName" in the new drawing. When a DXF with a external DWG is loaded, the layers with such names will be skipped since the character '|' is not allowed. In your case the layer names are valid so they are added.

I do not know how it works for your case, but if that layer name "0 @ 1" hides some kind of code, I do not understand why they are taking that route, since that string do not contain invalid characters, a user may have named a layer with that same name, and perhaps interfere with what it is trying to code. No idea.

Not every thing is supported and trying to load such objects may be end up partially or not loaded at all. At the moment I cannot do anything about it. Once again the official DXF documentation is pure crap, it leaves too many thing in the shadows. Why bother to create a drawing exchange format if you do not care to properly document it. And now I am beginning to rant.

haplokuon, 2020-11-04:
Sometimes AutoCad likes to write in the layer names, and perhaps in other places too, additional information. The details for your case I ignore them, but I know of a similar case. When a DWG is added to a another drawing as an external reference, layers of the attached DWG appears as "referenceName|layerName" in the new drawing. When a DXF with a external DWG is loaded, the layers with such names will be skipped since the character '|' is not allowed. In your case the layer names are valid so they are added.

I do not know how it works for your case, but if that layer name "0 @ 1" hides some kind of code, I do not understand why they are taking that route since, since that string do not contain invalid characters a user may have named a layer with that same name, and perhaps interfere with what it is trying to code. No idea.

Not every thing is supported and trying to load such objects may end up partially or not loaded at all. At the moment I cannot do anything about it. Once again the official DXF documentation is pure crap, it leaves too many thing in the shadows. Why bother to create a drawing exchange format if you do not care to properly document it. And now I am beginning to rant.

FilatovAV, 2020-11-10:
Here you can analyze changes in the DXF file.
If we look at the AcDbText text object, we will see that the following fields are assigned to It.

_«AcDbText
1001
AcadAnnotative
1000
AnnotativeData
1002
{
1070
     1
1070
     0
1002
}»_

The layer has the following entries

_«LAYER
  5
62F
102
{ACAD_XDICTIONARY
360
631
102
}
330
2
100
AcDbSymbolTableRecord
100
AcDbLayerTableRecord
  2
0 @ 96
 70
     0
 62
     7
  6
Continuous
370
    -3
390
F
347
4B8
1001
AcadAnnotativeDecomposition
1000
AnnotativeData
1002
{
1070
     1
1070
     1
1002
}
  0»
Changes in entity
«AcDbEntity
  8
0 @ 96»_

it was before

_«AcDbEntity
  8
0»_

haplokuon, 2020-11-16:
There is a lot more than meets the eye. It is not just a problem with the layer name, the extended data under the application registry named "AcadAnnotativeDecomposition" may also have something to do with it, and who knows what those 1 values below the code 1070 refer to. The layer also points to a dictionary, not shown in the pasted text, with handle 631 that may also hold more information related to this.

haplokuon, 2020-11-16:
There is a lot more than meets the eye. It is not just a problem with the layer name, the extended data under the application registry named "AcadAnnotativeDecomposition" may have something to do with it, and who knows what those 1 values below the code 1070 refer to. The layer also points to a dictionary, not shown in the pasted text, with handle 631 that may also hold more information related to this.