haplokuon/netDxf #232: Block - AttributeDefinitions - Value

rokolab, 2020-11-14:
If I iterare through AttributeDefinitions I get Tag and Prompt filled, but Value is empty? Is this missing when you read?

[image]

haplokuon, 2020-11-16:
That is the default value used for attributes when a new Insert entity is created. If it shows as an empty string it means than when the attribute definition of that block was created no value was assigned, so it may be normal.

Check the latest commit it contains some modifications that may affect you.

rokolab, 2020-11-16:
But I did insert a block in Autocad and assigned values. Try creating a block with one attribute in Autocad, then insert that block and assign value, then use this library and open autocad file, and you will see empty value.

rokolab, 2020-11-16:
I pulled latest version and it still does not work. See the example bellow:

[image]

[image]

haplokuon, 2020-11-16:
There seems to be a lot of confusion about this, I had to explain this many times even among AutoCad users. Do not confuse between inserts and blocks, an Insert is the actual entity that is a reference to a Block, a Block is the definition. You can have multiple inserts that references to the same block. An Insert contains Attributes, a Block contains AttributeDefinitions. This is the nomenclature of the DXF.

In AutoCad when you insert a block in a drawing you use the command "insert" which is the name of the entity, when you create a block you use the "block" command that as a shortcut also lets you to automatically create an insert in its place.

From your AutoCad screen capture you are showing the properties of a Block Reference, a block reference is an Insert not a Block.

rokolab, 2020-11-16:
Thank you, I found my atribute values under Inserts section. Now I understand.

MiguelG97, 2023-01-26:
Hi guys, I'm struggling as well with querying the blockreference attributes

[image]

is it ok to search for the document blocks and from there get their entities to find the inserts. Thus, I'll expect to find the attributes from inserts but the stringbuilder returns an empty string and the insert.Attributes.Count.ToString() also return a 0 collection.

`DxfDocument dxfFile = DxfDocument.Load(pathDXF);
foreach (netDxf.Blocks.Block block in dxfFile.Blocks){

      if (block.Name.Contains("318990"))
               foreach (netDxf.Entities.EntityObject ent in block.Entities)
                           if (ent is netDxf.Entities.Insert) 
                                    netDxf.Entities.Insert insert= ent as netDxf.Entities.Insert;
                                              if (insert.Block.Name.Contains("Window"))
                                                                foreach (var at in insert.Attributes) {
									stb1.AppendLine(at.Tag);
								}  
}`

I even tried to get the inserts right away from the document and same history
`			foreach (netDxf.Entities.Insert insert in dxfFile.Inserts) {
				
				stb1.AppendLine(insert.Attributes.Count.ToString());
			} `
Could you guys point me to the right direction? 
Thanks in advance!

MiguelG97, 2023-01-27:
Sorry for bringing this topic back to live, the problem was with the Software Revit. I'm using it to export dwg files in dxf format. However, it's not mapping the attributes with the entities. I might request the team to work directly with dxf format.
All the best,
Miguel G.