haplokuon/netDxf #235: font STANDARD

devel0, 2020-11-30:
I created a simple test file using librecad, containing a text without specifying a font name,

DxfDocument.Load fails returning null at [this point](https://github.com/haplokuon/netDxf/blob/8e9b81431f347123ab5da9e8a6f0ab6693d56ac6/netDxf/Tables/TextStyle.cs#L110) where font var = "STANDARD".

repro : [test file](https://drive.google.com/uc?export=download&id=1shvqkWGfuYbKS5z9BnlpE0X7fFwkk2nJ)

haplokuon, 2020-12-01:
Your DXF contains a TextStyle, in this case, the one name "STANDARD" that points to a font file named "txt". This is not a valid file name it should have included its extension. Also you are suffering [this](https://github.com/haplokuon/netDxf/issues/228) same issue. 

In one of the latest commits I have added several workarounds for buggy files such as yours, try it. For your standard text style the extension SHX will be automatically added font names without it, keep in mind that the actual existence of such a file is not checked by netDxf.

 I guess the LibreCAD DXF exported is kind of muddy...

devel0, 2020-12-01:
I confirm that what you said, sorry for dupe issue.
As remedy  I found that using QCad the saved dxf will opened without any problem.

note: the error showd up to me only using the source from a fork netDXF.standard because of the #32 but I suggest you to change from net framework to net 5.0 that could resolve both issue even I not already tried, in case I'll do I let you know results. Thanks for your useful project.

closed 2020-12-01 by devel0