haplokuon/netDxf #236: Layer of entities not detected

Corazzk, 2020-12-09:
Hi all,

please take a look at the DXF in attach. When this DXF is imported the layer to which each entity belongs is not recognized. All entities are on layer 0. This happens because in the header of the entity some information are stored in an unexpected order:

  0
TEXT
  5
1326
**100
AcDbEntity
100
AcDbText**
  8
Numery_dzialek
 62
256
370
-1
  6
BYLAYER

Instead of:

  0
TEXT
  5
1326
330
1F
**100
AcDbEntity**
  8
Numery_dzialek
**100
AcDbText**

As you can see in the first case layer is written after the second 100 tag and this cause the problem in ReadEntity procedure.  

It would be possible to fix it ?

Here the DXF:

[ex_02_MZ_DXF.zip](https://github.com/haplokuon/netDxf/files/5666247/ex_02_MZ_DXF.zip)

Thank you in advance.

Best regards.

Keven Corazza

haplokuon, 2020-12-11:
I am aware of this issue, but I am not going to fix this. The program that generated that file should be the one responsible of fixing this. Under the subclass marker AcDbEntity should go all related properties that are common to all entities, such as, the layer, the linetype, the lineweight, the color, etc.

Corazzk, 2020-12-11:
Thank you for your answer.

Yes I know what you mean but, considering that AutoCAD is able to read this file in the right way, maybe you can also try to support it.

Thanks. Best regards.