haplokuon/netDxf #238: Dimension in a plane different form XY

subGm, 2020-12-10:
Hello,
I would like to add linear dimensions to some graphic objects placed in a plane different from XY (in this case parallel to XY with a Z offset).
I would like to add linear dimensions to these objects, but the LinearDimension constructors only take Vector2 as paramters.
Is there a way to solve this issue?
Thanks

subGm, 2020-12-10:
I used this workaround:
1. Instantiated **LinearDimension** object with the constructor that takes **Vector2**.
2. Rotated and translated the object with **TransformBy** method.

haplokuon, 2020-12-11:
The Normal and Elevation properties defines the construction plane of the Dimension entity. The Normal defines the local plane and the Elevation the distance along the local Z axis. The Vector2 parameters are coordinates in the local plane.

You can use the TransformBy method since it changes the local plane of the dimension, but it is a lot more costly, it not only applies rotations but scales and translations too.

subGm, 2020-12-11:
Thanks for your support.

closed 2020-12-11 by subGm