haplokuon/netDxf #246: How can read the DXF and draw the shapes and display using dot net

sachinpawar13, 2021-01-03:
private async void button1_Click(object sender, System.EventArgs e)
        {
            if (openFileDialog1.ShowDialog(this) != DialogResult.OK) return;
            if (openFileDialog1.FileName != null)
            { 
                DxfDocument loaded = DxfDocument.Load(openFileDialog1.FileName);
                int counter = 1;
                foreach (netDxf.Entities.Image image in loaded.Images)
                {
                    DxfDocument doc = new DxfDocument();
                    netDxf.Entities.Image copy = (netDxf.Entities.Image)image.Clone();
                    doc.AddEntity(copy);
                    string fileName = string.Format("Image{0}.dxf", counter++);
                    doc.Save(fileName);
                    // load file
                    doc = DxfDocument.Load(fileName);
                    doc.DrawingVariables.AcadVer = DxfVersion.AutoCad2010;                 
                }
             }

haplokuon, 2021-01-04:
netDxf do not provide any kind of functionality to display the contents of the a DXF, you will need to code it yourself. You could have different options are your disposal, you could use the API the NET Framework supply, [OpenTk](https://github.com/opentk/opentk) for OpenGL, or , [SlimDX](https://github.com/SlimDX) for DirectX programming, this last one seems not very well supported kinda abandoned. If you want a more high level engine, you also have your options [Godot](https://godotengine.org/), [NeoAxis](https://www.neoaxis.com/), or even [Unity](https://unity.com/).

There are others just search a little bit, you have many options, you could even use C++. This question is too broad.

sachinpawar13, 2021-01-04:
> netDxf do not provide any kind of functionality to display the contents of the a DXF, you will need to code it yourself. You could have different options are your disposal, you could use the API the [...]

closed 2021-01-04 by sachinpawar13