haplokuon/netDxf #247: there is a bug in project

Han-gege, 2021-01-07:
if (!(Math.Abs(this.knots[i + p] - this.knots[i]) <= double.Epsilon))

The return result is uncertain
sometimes The return result is true
sometimes The return result is false

Han-gege, 2021-01-08:
[image]
[image]

haplokuon, 2021-01-17:
The spline method you are showing is used by the "PolygonalVertexes" method. Did you have any problems while using that method, for example when converting a spline to a polyline?. That piece of code was provided by another user very long time ago, and I didn't have problems with it. That conditional is there to avoid a division by zero, personally I would use the MathHelper.IsZero method, it use a much larger epsilon value than double.epsilon.

What you are showing might be the result of a comparison rounding error for very low numbers, but as long as you didn't get a division by zero exception everything should be fine. Post a piece of code with the problem that I can actually reproduce, a screenshot of some code has very little use, I cannot copy-paste a text from it.