haplokuon/netDxf #255: How to add arc through LwPolyline

itcasth, 2021-02-03:
Hello author, how to add arc through lwpolyline?

itcasth, 2021-02-03:
I know that I can draw an arc through bulge, but I need to know the angle of the start point and the angle of the end point. How can I draw an arc through the coordinates of the start point, the coordinates of the end point and the center of the circle

haplokuon, 2021-02-03:
Take a look at the method Explode of the LwPolyline it will show you how it is done, this method converts a lwpolyline into a set of lines an arcs. Also you can check the method PolygonalVertexes.

itcasth, 2021-02-04:
Thank you very much！ I found that I can add arcs. Can I add multiple lines and arcs at the same time with one lwpolyline. Not through multiple lwpolylines

itcasth, 2021-02-04:
Thanks! I've solved it

closed 2021-02-04 by itcasth