haplokuon/netDxf #266: Maximum file size to load

tyrgin, 2021-03-04:
Hello @haplokuon and thank you for your work!

I was wondering is there any limitation of loaded file size? 
I have some times ugly heavy documents (which mostly are converted from PDF into millions of polylines) which do not load.

Best regards,
Valentin

haplokuon, 2021-03-04:
Yes, memory is a limit and the maximum number of items in the list and dictionaries that holds the data also has limits. I tried with drawings with 5 millions lines (1GB text DXF files) and they didn't give problems; but not all entities are the same. For example, using Polyline entities (aka 3d polyline) is not the same as using LwPolyline entities (2d polylines), the latter are much more memory friendly. Download the code and try to open your files in Debug mode to know where it fails.

The dictionary where the handles of the DXF objects are stored may be the first to fail, if you are using 3d polylines in your drawing every single vertex consumes a handle and if you have millions of polylines, just a handful of vertexes in them can make the dictionary to grow very fast. If this is your problems there could be a solution around that, like storing them somewhere else or not storing them at all. Handles are often used in a DXF as a means to reference other objects, but handles of a 3d polyline vertex not really, at least at the moment I don't know why such handles might be needed.

It is also possible that your files fail for some other reason not related to memory limits.

tyrgin, 2021-03-04:
okay, will try, thank you!

tyrgin, 2021-03-04:
Well, I don't get it yet, but running you code directly I don't encounter any issues...
more than that, loading dxf (45MB) in my code (simple call `DxfDocument.Load()`) return in 130 seconds and if I try load 200MB it returns never. Your code returns document of 550MB file in 45 seconds...
Any ideas? I will dig further tomorrow

haplokuon, 2021-03-05:
You were using a very old version of the library around the time when I introduced, unwillingly, a nasty and unneeded check that made the loading process painfully slow, it was fixed long ago. You can read about it [here](https://github.com/haplokuon/netDxf/pull/104). The file would eventually finish but probably instead of waiting a minute for a large file you would have to wait an infuriating hour, who knows I didn't have the patience to test it.

Next time check here at Github if there are any updates, I do not upload anything outside here.

tyrgin, 2021-03-06:
I see now!
Thank you very much!

closed 2021-03-06 by tyrgin