haplokuon/netDxf #271: How to split entities

subGm, 2021-03-31:
Hello,

I'm wondering if with netDxf it is possible to split a 2d entity using another entity.

Something like the example below, where the red polyline is used to split the two circles:

[image]

Thanks

haplokuon, 2021-04-01:
No. The focus of this library is reading and writing DXF files not to solve geometric problems, but this is an easy problem. You just need to solve the intersection between a line segment and a circumference, there is plenty of information around the internet on how to do it. Next from the intersection point you can calculate the start and end angles of the arc, and since you already know the center and the radius you can create the new Arc entities.

subGm, 2021-04-01:
Thanks for your reply.
You are right, my example is an easy to solve, geometric problem. But it was only a simple example to clarify my question.
Actually, my final goal is different. If I have a block (made of many different entities), I would like to put a rectangle on the block, and then setting a layer for the part of the block inside of the rectangle, and another layer for the outside.