haplokuon/netDxf #274: Handling LEADER's inversed "Horizontal" direction

watanabelikewelltech, 2021-04-12:
My cad file has some leaders which has horizontal direction of  "X=  -1.0000  Y=   0.0000  Z=   0.0000".
which netDxf.Entities.Leader shows x-axis mirrored.

This is the dxf's leader part.
`LEADER
  5
4B5BB
330
2
100
AcDbEntity
  8
Defpoints
100
AcDbLeader
  3
HG-SAMPLE
 76
     3
 10
2257273.242409303
 20
1357426.800043135
 30
0.0
 10
2262539.567050831
 20
1370603.10977628
 30
0.0
 10
2271363.928777016
 20
1370603.10977628
 30
0.0
211
-1.0
221
0.0
231
0.0
210
0.0
220
0.0
230
-1.0
1001
ACAD
1000
DSTYLE
1002
{
1070
     6
1000
OPEN
1070
    77
1070
     0
1070
   147
1040
0.5
1070
   343
1005
23
1002
}
  0
TEXT
  5
4B5BC
330
2
100
AcDbEntity`

It says 211 -1.0 which appears horizontal direction inversed.

This the loaded netDxf.Entities.Leader.

-- | -- | -- | --
◢ | Vertexes | Count = 3 | System.Collections.Generic.List<netDxf.Vector2>
  | ▶ [0] | {-2257273.242409303, 1357426.800043135} | netDxf.Vector2
  | ▶ [1] | {-2262539.567050831, 1370603.10977628} | netDxf.Vector2
  | ▶ [2] | {-2271363.928777016, 1370603.10977628} | netDxf.Vector2

This is the vertex appears on Autodesk DWG TrueView 2021.

X=2257273.2424  Y=1357426.8000  Z=   0.0000
X=2262539.5671  Y=1370603.1098  Z=   0.0000
X=2271363.9288  Y=1370603.1098  Z=   0.0000

I think netDxf is right and autodesk is wrong as it seems to ignore horizontal direction
but still there's no way I can fix that mirroed LEADERs' postions.

haplokuon, 2021-04-15:
Upload the full DXF file with just that leader. Looking at how the Leader is read from the DXF looks like that I forgot to read the codes 211, 221, and 231. At the moment I am just defining the horizontal vector as the direction of the last two vertices of the leader, but there may be special cases where you want to place the leader's annotation with a different direction, and at the moment there is no parameter in netDxf for that.

watanabelikewelltech, 2021-04-16:
The attached dxf contains 9 leaders where 6 are normal and the other 3 (Handle=2E5,2E8,2EB) have -1.0 value.

[invertedLeadersSample.zip](https://github.com/haplokuon/netDxf/files/6321901/invertedLeadersSample.zip)

haplokuon, 2021-04-19:
I took a look at your file and it is not the same as in your first post. I do not see any graphical difference between your original file and the same one saved with netDxf. All of them are leaders with an arrow end pointing to the right when looking down to the XY plane. If you try to superimpose both drawing, their coordinates are exactly the same. So, I do not see what is the issue here or this "mirrored", do not worry to much on how the DXF stores its stuff, sometimes I interpreted them in a different way. I still do not know what this horizontal vector is for, besides adding an unnecessary complication.

What I see is a difference in the Normal values, your "mirrored" leaders have the normal vector as (0,0,-1) and the others (0,0,1). Keep in mind that the leader vertices are stored in local coordinates, it is in the documentation, that is why you see the change in the sign for the X-coordinate. And that is why is important to present full pieces of working code or full DXF files, the evil is in the details.

watanabelikewelltech, 2021-04-19:
I noticed I was wrong about mentioning "mirrored", it's rather "x becomes negative".

I followed the advice that Normal.Z  and
the code below now works fine for both (0,0,1) and (0,0,-1).
Thank you.

```csharp
var normalized = MathHelper.ArbitraryAxis(obj.Normal).Transpose();
var invX = normalized.M11;
var invY = normalized.M22;
//convert Leader to NetTopologySuite LinesSring
var line = factory.CreateLineString(obj.Vertexes.Select(x => new Coordinate(x.X * invX, x.Y * invY)).ToArray());
`