haplokuon/netDxf #275: Cannot rotate an image

ponahoum, 2021-04-13:
Hi,

I successfully managed to generate a DXF with images, my images are properly positioned and sized. 

The issue is that my images are always in the XY plane. I want to rotate these images so that they be in the XZ plane for example, but the rotation doesn't seem to be an option in the Image constructor:
` public Image(ImageDefinition imageDefinition, Vector3 position, Vector2 size);`

I tried to play with rotation matrix, but without success. Any clue ? :)

ponahoum, 2021-04-14:
Alright I made it ! 

`
Image imageDxf = new Image(def, new netDxf.Vector3(0,0,0), imageWorldSize);
imageDxf.TransformBy(Matrix3.RotationX(Mathf.Deg2Rad*90), netDxf.Vector3.Zero);
imageDxf.TransformBy(Matrix3.Identity, imageBottomCorner);
`

haplokuon, 2021-04-15:
You need to set the Normal and Rotation properties to define the orientation of an entity, specially for pure 2d entities like Image, Text, Circle, Arc,... The Normal defines the normal of the plane where the entity is placed, in your case, (0,1,0); the rotation represents the angle around the local Z axis. While using the TransformBy methods will work, they modify the normal and rotation according to the supplied matrix, it is a lot easier for simple case to just set the Normal directly.

ponahoum, 2021-04-15:
@haplokuon good point, it's much simpler this way!

closed 2021-04-15 by ponahoum