haplokuon/netDxf #276: About write large dxf file

doublewei6, 2021-04-17:
Hi, 
I want to write a large dxf file, for example larger than 4GB, I want to write data using stream, how to do that?
Thanks

haplokuon, 2021-04-18:
There are two samples in the TestDxfDocument project, look for the methods called "ReadWriteFromStream" and "ReadWriteFromStream2". Those methods use the FileStream and MemoryStream, to work the stream needs to have the property Position accessible, and not all seems to allow it, read about it [here](https://github.com/haplokuon/netDxf/issues/273).

doublewei6, 2021-04-19:
Hi, I want to write large file like following, however, i only found dxf.Save(memoryStream), and I do not know how to write stream in for circle
create file 
for
{
data to stream;
write stream
stream.flush()
}
close stream;
Thanks!

doublewei6, 2021-04-19:
And please update VS NuGet which has two version, netDxf v2.2.0.1 and nerDxf.netstandard v2.4.0

haplokuon, 2021-04-19:
You cannot do that, the information stored in the DxfDocument must be saved in one go. It is not like you can easily save a single entity, the information needed to represent a single, for example, circle is stored in many places in the DXF, you need to take care of handles, layers, linetypes, ... and there are entities a lot more complex than a circle.

I cannot update anything at NuGet, I do not upload anything there. So you better ask the person who did that, I tell him/her to delete it if they do not want to keep it up to date. The version 2.2.0 is very old, it would be much better if you download the latest code and compile it yourself. The latest iteration is a multitarget project, predefined frameworks for Net Framework 4.5, Net Standard 2.1, Net Core 3.1, and NET 5.0. You can easily change the framework target to suite your needs.

doublewei6, 2021-04-25:
And how to get border coordinary, （left_down.X,left_down.Y,width,height）

haplokuon, 2021-04-25:
Do you mean [this](https://github.com/haplokuon/netDxf/issues/27)?

doublewei6, 2021-04-28:
extmin and extmax sometimes not right,now I just read each entity and calculate its border,this choice is not slow