haplokuon/netDxf #277: DXF defective if it contains an Ellipse

MatthiasSchilder, 2021-04-21:
In one of our programs, I noticed the created dxf seems to be defetive sometimes. AutoCAD (2020, 2021) can't open those dxfs. I've also tried to open the files with QGIS 3.10. QGIS seems to be a little bit more tolerant, sometimes it is able to open those files, sometimes not.

To me, it looks like it always happens, when the dxf contains Ellipses. See my minimal working example.
It creates two dxfs: one containing only a Line. The resulting dxf can be opened with AutoCAD. The other one only contains an Ellipse with Color red. This dxf cannot be opened with AutoCAD. I also attached both dxfs. I don't see any exceptions, return value of doc.Save(..) is always true.

Tried it with versions netDxf 2.2.1 and netDxf.standard 2.4.0, both from nuget.org.

Is this an issue with netDxf or with the dxf-consuming software?

```csharp
static void Main(string[] args)
{
	var docDefective = new DxfDocument(DxfVersion.AutoCad2013);
	Ellipse ellipse = new Ellipse(new Vector2(105.4387, 321.6599), 0.3, 6);
	ellipse.Color = new AciColor(0, 0, 255);
	docDefective.AddEntity(ellipse);
	var result = docDefective.Save($"c:\\users\\{Environment.UserName}\\desktop\\defective.dxf");
	Console.WriteLine(result);

	var docOk = new DxfDocument(DxfVersion.AutoCad2013);
	Line l = new Line(new Vector2(105.4387, 321.6599), new Vector2(205.4387, 321.6599));
	docOk.AddEntity(l);
	result = docOk.Save($"c:\\users\\{Environment.UserName}\\desktop\\ok.dxf");
	Console.WriteLine(result);
}
```
[dxfs.zip](https://github.com/haplokuon/netDxf/files/6349573/dxfs.zip)

MatthiasSchilder, 2021-04-21:
This is what AutoCAD 2021 (german) says when loading defective.dxf:

> Folgender Fehler wurde beim Lesen
> in ELLIPSE beginnend bei Zeile 1068 ermittelt:
> Unzulässige Daten für Ellipsendefinition
> Ungültige oder unvollständige DXF-Eingabe -- Zeichnung abgebrochen.

Translated: 

> The following error occurred while reading
> determined in ELLIPSE starting at line 1068:
> Invalid data for ellipse definition
> Invalid or incomplete DXF input - drawing aborted.

haplokuon, 2021-04-23:
The Ellipse entity works fine, but the major axis, as its names implies, must be greater than the minor axis.

MatthiasSchilder, 2021-04-23:
Thanks.
Would you mind throwing some kind of ArgumentException in this case?