haplokuon/netDxf #280: Enumerate layer entities

YaroslavKormushyn, 2021-04-23:
Hello! First off, great library!

I want to enumerate entities of a layer. There's `Layers` on the `DxfDocument`, but the layers don't contain their entities. All of the entities are rather distributed between different collections.

Is there a way to get all of the entities without going through all of their collections?

What I'm looking for is a way to get all different file geometry types (`ARC`, `POLYLINE`, etc.)

haplokuon, 2021-04-24:
Not exactly, but you can use the method GetReferences(tableObjectName), that is inherited by all table object collections, to retrieve the all DxfObjects that reference a particular table object. In the case of the Layers collection it returns all attributes, attribute definitions and entities that have a reference to a specific layer. This includes entities of a user defined block and also entities in blocks associated with every layout.

In the DXF structure every entity, in one way or the other, belongs to a block. The ModelSpace layout, for example, has an associated block called "*Model_Space" where the entities are stored. All those Arcs, Circles, Lines,... collections are no more that shortcuts to the block where they are stored. In the DxfDocument project there are sample about all these.

YaroslavKormushyn, 2022-02-11:
Thanks! This answers my question.

closed 2022-02-11 by YaroslavKormushyn