haplokuon/netDxf #282: Issue with BeforeAddItem in ObservableDictionary

p-skakun, 2021-04-27:
Hi!

I started to face issues with loading some .dxf files after upgrading to latest GitHub version from the one from NuGet package
It looks like the reason is in the unhandled exception raised from the next line: https://github.com/haplokuon/netDxf/blob/5060604146f423e74c652a5695b650f10e8acca1/netDxf/Collections/ObservableDictionary.cs#L172
[An example](https://drive.google.com/file/d/1my94qgZwdBPKeZf0Xp4hp_ws-joP8lrX/view?usp=sharing)
The handler actually leads to `LayerState.Properties_BeforeAddItem`, where event key is not in the list of layers.
It seem to be the layer for AcDbXrecord, but it's not in the layers list.

Could you share some brief reasoning behind this, please? Also it would be great if you could point me to some workarounds.

Thanks

haplokuon, 2021-05-02:
Check the latest update, your file should be working now. The issue was related with the layer states of the drawing, they contained multiple references to layers not present in the drawing. When such references comes from a DXF those layer states will not be added, but when they come from an external LAS file layer states, that reference a layer that does not exist in the document, they will create a new layer, this way such file can be used as layer templates.

p-skakun, 2021-05-03:
It works! Thank you for the explanation and the fix!

closed 2021-05-03 by p-skakun