haplokuon/netDxf #284: Clone annotation causes increase its number of vertices

kyosukeSakurai, 2021-05-13:
Clone annotation causes increase its number of vertices.
I suppose it's a bug.
I expect that the number of vertices will not increase even if I clone it.
The following is a reproduction sample code and dxf files.
note: To try the sample code, create a folder named 'tmp' directly under the C drive in advance to store the sample dxf file 'BeforeCloning'. 
[image]

[tmp.zip](https://github.com/haplokuon/netDxf/files/6471654/tmp.zip)
### 

Sample code: 
static void Main(string[] args)	
 {
            DxfDocument dxffile = new DxfDocument();	
            string debugfile = "C:\\tmp\\BeforeCloning.dxf";	
            DxfDocument a = DxfDocument.Load(debugfile);	
            var leaders = a.Leaders.ToList();	
            foreach (var leader in leaders)	
            {	
                var d = leader.Clone();	
                a.AddEntity(d as EntityObject);	
            }	
            a.Save(@"C:\tmp\AfterCloning.dxf");	
        }	
------------------------------------------------------------	
HasHookline seems to be increasing vertex, but I don't understand this specifications (why insert a vertex that I don't seem to need ?).	
	
 public bool HasHookline	
        {	
            get { return this.hasHookline; }	
            set	
            {	
                if (this.hasHookline != value)	
                {	
                    if (value)	
                        this.vertexes.Insert(this.vertexes.Count - 1, this.CalculateHookLine());	
                    else	
                        this.vertexes.RemoveAt(this.vertexes.Count - 2);	
                }	
                this.hasHookline = value;	
            }	
        }

kyosukeSakurai, 2021-05-14:
I tentatively changed the code so as not to increase the vertex. 
After the changes:

 ```
public bool HasHookline

        {
            get { return this.hasHookline; }
            set
            {
                if (this.hasHookline != value)
                {
                    if (value)
                        if (this.vertexes.Count >= 3)
                        {
                            this.vertexes[this.vertexes.Count - 1] =  this.CalculateHookLine();
                        }
                        else
                        {
                            this.vertexes.Insert(this.vertexes.Count - 1, this.CalculateHookLine());
                        }
                    else
                        this.vertexes.RemoveAt(this.vertexes.Count - 2);
                }
                this.hasHookline = value;
            }
        }
```
[image]

However, the shape of  leader is not maintained. 								
I expect the number of vertices should not increase and the shape of leader should be maintained. 								
Please consider it.

haplokuon, 2021-05-23:
I have updated the code, now the shape of the leader should be preserved. The root of the problem was that some of the manually edited properties of the leader were overwritten by default values when loading it from the DXF.

kyosukeSakurai, 2021-06-14:
Certainly vertex seems to have stopped increasing. However, as I told you in advance, the direction of the leader line is reversed. Is it possible to prevent it from reversing?

haplokuon, 2021-06-20:
Nothing gets inverted when cloning your original leader. If you are talking about your second set of images, well... weird thing may happen when you modify the code, that's your responsibility. Or show an example were it happens using an unmodified version of the code.

kyosukeSakurai, 2021-06-21:
I registered in new issue about the leader of annotation.
See #293.

haplokuon, 2021-06-28:
I will keep answering here since all this is related.

Do you use the QLeader command in AutoCad to generate your leaders instead of the Leader command? The QLeader has the bad habit of not using the attachment point of align the leader text, instead it places the attachment point somewhere and it moves the text to the correct position. Since I cannot calculate the length of a text, I always use the alignment point to place the text. The only solution for this is just save the parameters from the DXF as they are and not try to recalculate anything. It will work but the side effect is that if you try to call the Update method of the leader, it might have unexpected results.

haplokuon, 2021-10-17:
Read the [changelog.txt](https://github.com/haplokuon/netDxf/blob/master/doc/Changelog.txt). This is a change that I did sometime ago, now you can find them under DxfDocument.Entities.Leaders

closed 2022-05-30 by haplokuon