haplokuon/netDxf #286: DXF saved in AutoCAD 2000 format cannot be loaded

kyosukeSakurai, 2021-05-28:
DXF saved in AutoCAD 2013 format can be loaded with the Load method of the DxfDocument class.
However, the latest netDXF throws an exception when trying to load DXF saved in AutoCAD 2000 format with Load.
We want to load DXF saved in AutoCAD 2000 format, but we are hav[tmp.zip](https://github.com/haplokuon/netDxf/files/6560793/tmp.zip)ng trouble loading it.
I want you to be able to load DXF in AutoCAD 2000 format.
Earlier versions (probably netDxf 2.4.0) were able to load.

The following is a reproduction sample code and dxf files.

"AC1015.dxf" is the DXF saved in AutoCAD 2000 format, and "AC1027.dxf" is the DXF saved in AutoCAD 2013 format.
note: To try the sample code, create a folder named 'tmp' directly under the C drive in advance to store the sample dxf file.

[tmp.zip](https://github.com/haplokuon/netDxf/files/6560798/tmp.zip)

Sample code:
```
static void Main(string[] args)	
{
	// Successes loading.
	var dxfDocument = netDxf.DxfDocument.Load(@"C:\tmp\AC1027.dxf");
	// Throws exception.
	var dxfDocument = netDxf.DxfDocument.Load(@"C:\tmp\AC1015.dxf");
}
```

kyosukeSakurai, 2021-06-06:
Isn't it affected by the fact that the Shift-JIS (character code) processing code has been deleted in the latest version? 

DxfReader.cs
public DxfDocument Read(Stream stream, IEnumerable<string> supportFolders)

[image]

haplokuon, 2021-06-20:
It should work now once again.