haplokuon/netDxf #287: Arc startAngle or EndAngle Issue

petercaogx, 2021-06-03:
I am working on exporting AUTOCAD dxf file to svg file, and most of function worked perfect, but Arc startAngle or endAngle,
I tried to correct it in two ways if the object is Arc I will 
if (l.EndAngle > l.StartAngle)
            {
                startAngle = (float)(CIRCLE - l.EndAngle);
                stopAngle = (float)(l.EndAngle - l.StartAngle);
            }
            else
            {
                startAngle = (float)(CIRCLE - l.EndAngle);
                stopAngle = (float)(CIRCLE - l.StartAngle + l.EndAngle);
            }
if the object is Ellipse I will
float startAngle = 0.0f;
                float stopAngle = 0.0f;
                if (l.EndAngle > l.StartAngle)
                {
                    startAngle = (float)(225 - l.EndAngle);
                    stopAngle = (float)(l.EndAngle - l.StartAngle);
                }
                else
                {
                    startAngle = (float)(225 - l.EndAngle);
                    stopAngle = (float)(360 - l.StartAngle + l.EndAngle);
                }
Any one can help on this issue?

Thanks

[image]
[image]

petercaogx, 2021-06-03:
Sorry, The CIRCLE is a const = 360

haplokuon, 2021-06-10:
You still need to explain what are you trying to do and what that piece of code represents and what are you trying to achieve with it. Imagine that I have no idea about the SVG format and its requirements.