haplokuon/netDxf #292: AcDbCircle with extrusion direction

HaraldHoerst, 2021-06-08:
Hello Daniel,

I have received two versions of an AcDbCircle Object. When using the Circle.ToPolyline() Function, I get a proper Polyline for the first version, bit none for the second. As far as I understand, the difference are the parameter 210-230, which I understand as the "Extrusion direction", parameter 10, the negated X-CenterPoint, and the parameters 50,51, the start-/stop-angles.

Why is the second version not working?

Regards,

Harald Hörst

working version 1:-----------
  100
AcDbCircle
  10
298505.2428
  20
108748.5544
  30
0.0
  40
1000.0
  100
AcDbArc
  50
270.0
  51
320.5857633
  0
ARC
  5
31
  330
13
------------------------------

failing version 2:--------------
  100
AcDbCircle
  10
-298505.2375
  20
108748.548
  30
0.0
  40
1000.0
  210
0.0
  220
0.0
  230
-1.0
  100
AcDbArc
  50
219.4140517
  51
270.1987283
  0
LINE
  5
25
  330
13
------------------------------

haplokuon, 2021-06-10:
Do not post pieces of a DXF, I need the whole file, and a way to reproduce what are you doing.

HaraldHoerst, 2021-06-11:
Hi,

please find attached the two relevant files:
[Ver2_ARC_neg_NormalZ.dxf.txt](https://github.com/haplokuon/netDxf/files/6636009/Ver2_ARC_neg_NormalZ.dxf.txt)
[Ver1_ARC_pos_NormalZ.dxf.txt](https://github.com/haplokuon/netDxf/files/6636010/Ver1_ARC_pos_NormalZ.dxf.txt)

haplokuon, 2021-06-13:
I do not see any issues converting the arcs in your drawing into LwPolylines. My guess is that you are reading the LwPolyline vertexes as if they were in world coordinates, they are in local/object coordinates, it is in the documentation. In the MathHelper class they are several methods to handle the transformation of vertexes to and from WCS and OCS. Also check the MathHelper method "public static Matrix3 ArbitraryAxis(Vector3 zAxis)".