haplokuon/netDxf #293: The coordinates of the configuration points of the annotation become invalid

kyosukeSakurai, 2021-06-21:
The coordinate values of the configuration points of the annotation become invalid. 
It seems that the cause is the netDXF version upgrade, so please confirm it.

I send you sample codes to check the difference between the old and new versions. 
    New version: ConsoleAppNewNetDxfLoad
    Previous version: ConsoleAppOldNetDxfLoad
    
[Sample.zip](https://github.com/haplokuon/netDxf/files/6686556/Sample.zip)

     Input data: C:\tmp\netDxfInput.dxf
     Output data: C:\tmp\newNetDxfOutput.dxf, C:\tmp\oldNetDxfOutput.dxf
     Output destination: Console, VisualStudio Output Window

The execution result should be as follows:
[image]

I think the following processing is invalid.
Line 894 on Leader.cs

CalculateHookLine processing is significantly different from past versions I am recalculating from this.Direction, but since Annotation does not exist at the time of loading, this.Direction is always x = 1, y = 0 and HookLine is calculated with invalid coordinates.

haplokuon, 2021-06-28:
I will continue the discussion in the [older issue](https://github.com/haplokuon/netDxf/issues/284) since the origin of those problems seems to be the same.

kyosukeSakurai, 2021-06-29:
Load, update of leader and save seems to cause other inconsistencies.
[image]

I think the following processing is needed to be fixed. 
(Since the update position of Hook at the time of update is inappropriate, do the same as HasHookLine.)
Line 551 on Leader.cs

```
            if (this.hasHookline)
            {
            	// Remove the 3rd point and insert the HookLine second.
                this.vertexes.RemoveAt(this.vertexes.Count - 2);
                this.vertexes.Insert(this.vertexes.Count - 1, this.CalculateHookLine());
                //Vector2 vertex = this.CalculateHookLine();
                //this.vertexes[this.vertexes.Count - 2] = vertex;
            }
```