haplokuon/netDxf #294: The direction of the annotation leader is unintentionally reversed

kyosukeSakurai, 2021-06-22:
The direction of the annotation leader is unintentionally reversed

Load & Save the sample data (SampleX.dxf). 
In the previous netDxf version, Leader group code '211' becomes '1'. 
When opened with a viewer such as AutoCAD, the correct display is as shown below. 
[image]

In the new netDxf version, Leader group code '211' becomes '-1'. 
When opened with a viewer such as AutoCAD, the display is incorrect as shown below. 
[image]

I think it is a bug in the new version, so please check it. 

Ref. ) Leader group code 
https://help.autodesk.com/view/ACD/2020/ENU/?guid=GUID-396B2369-F89F-47D7-8223-8B7FB794F9F3

haplokuon, 2021-06-28:
I will continue the discussion in the [older issue](https://github.com/haplokuon/netDxf/issues/284) since the origin of those problems seems to be the same.

kyosukeSakurai, 2021-06-29:
If issues #293 and #294 are solved, we can see the following issues.

If the annotation leader line has an acute angle, Load and save will make it impossible to reproduce the acute angle.
[image]

Sample code:
```
var dxfDoc = DxfDocument.Load( filename );
dxfDoc.Save( filename );
```
[SAMPLE_C.zip](https://github.com/haplokuon/netDxf/files/6733585/SAMPLE_C.zip)

I think the following processing is needed to be fixed.
Line 4511 on DXFReader.cs

```
            if (hasHookline && wcsVertexes.Count >= 3)
            {
		// When 1st, 2nd and 3rd points make up an acute angle, not to delete the 2nd point 
               //  because it need to be restored. See 'WriteLeader' on 'DxfWriter.cs'.
                var x1 = wcsVertexes[2].X - wcsVertexes[1].X;
                var x2 = wcsVertexes[1].X - wcsVertexes[0].X;
                if ((x1 * x2) > 0)
                {
                    wcsVertexes.RemoveAt(wcsVertexes.Count - 2);
                }
            }
```
Line 364 on Leader.cs
```
              // Recalculate and add Hookline only if there are two vertices.
              if (this.vertexes.Count == 2)
              {
                      this.vertexes.Insert(this.vertexes.Count - 1, this.CalculateHookLine());
              }
```
[FixedCode.zip](https://github.com/haplokuon/netDxf/files/6733618/FixedCode.zip)