haplokuon/netDxf #299: Edit MText entity; need general advice

fjod, 2021-07-05:
Hello!
I need to write plugin to some CAD system. It should edit some MText values. I managed to find them inside document and read; but when I try to write Contents fields and save doc, it becomes corrupted. When I try to use InplaceTextEditor to save changes, it looks like it throws some exception and my code exits after first entity.
I worked a little with netDxf and know that it's possible to edit MText there; mb you know how I can edit them using CAD objects?

haplokuon, 2021-07-07:
It is impossible to give any meaningful answer to such a broad question. This library is only to load, edit, and save DXF files. It is completely independent of any CAD software and does not require any of them. If you need to edit or create MText entities with netDxf check its properties and methods and check what they do, but what is wrong with your plugin for an unidentified CAD software I cannot help.

fjod, 2021-07-08:
> It is impossible to give any meaningful answer to such a broad question. This library is only to load, edit, and save DXF files. It is completely independent of any CAD software and does not require [...]

Sorry for this not relevant question. Turned out I did almost everything fine, I only had to switch between layers so CAD software redraws my changes. I thought it was about something with inner dxf structure.

closed 2021-07-08 by fjod