haplokuon/netDxf #305: About comment in output file

is0263rf, 2021-07-15:
This library currently licensed under MIT license. But, default comment in output file says LGPL. Is there some kind of meaning to that?

https://github.com/haplokuon/netDxf/blob/e1b64e9f19f9e0fc7f72ed7b7b746f2682849561/netDxf/DxfDocument.cs#L164

haplokuon, 2021-08-07:
I have changed the license a few months ago and I forgot to change that. Don't worry about it, you can easily edit, delete, and add your own comments.

is0263rf, 2021-08-08:
Thank you. I sent pull request.

haplokuon, 2021-08-09:
Thank you for your pull request, it won't be necessary, I already did the changes in my working copy. More programs should make use of the comments at least to identify its origin, even though they are not used in binary DXF files.

is0263rf, 2021-08-10:
I see. My question was solved.

closed 2021-08-10 by is0263rf