haplokuon/netDxf #317: Weird line type issue keeping dxf from being opened

donnyv, 2021-09-09:
Attached are 2 dxf files. They are exactly the same except the line type on layer  V-PROP-ASSESS has been changed to HIDDEN. For some reason when the line type is set to Wetlands. netDxf will not open it. I tried comparing the dxf file contents but its hard to make heads or tails out of that format. 

[LineTypeRemovedFor_V-PROP-ASSESS.zip](https://github.com/haplokuon/netDxf/files/7136917/LineTypeRemovedFor_V-PROP-ASSESS.zip)

haplokuon, 2021-09-11:
Your DXF files contains a lot of external referenced stuff. In your case, on my side, your file "0910_029_org.dxf" is the one giving me problems because it cannot find a required shape file. I may add a workaround for that, but I had no problems with the other file "0910_029_linetypechanged.dxf".

Try the latest code and the release build if you have not done so, yet.

donnyv, 2021-09-12:
0910_029_org.dxf is the one I had issues with. I'm using the latest release.

haplokuon, 2021-09-12:
I found a couple problems that will be fixed with the next update. One has to do with getting the information from the SHX file through the ShapeStyle class. The other one is related with linetypes defined in the layer states not found in the document linetype list, in your case due to the externally referenced drawings.

donnyv, 2021-09-12:
Unfortunately I don't have control of the drawing. I get them however they send it. I only need the actual line work in it. Thank you so much for looking at it.

donnyv, 2021-10-04:
I just ran across another dxf with the same issue but different layers. It could help you with testing.
[0909_021.zip](https://github.com/haplokuon/netDxf/files/7280601/0909_021.zip)

donnyv, 2021-10-06:
Funny thing is that this dxf came from a AutoCad export.

closed 2022-05-30 by haplokuon