haplokuon/netDxf #327: Quesiton about  Load 2 dxf files as Blocks

seppelhuang, 2021-10-30:
soure files

1.dxf

[image]

2.dxf

[image]

See The following codes:

           DxfDocument dxf = new DxfDocument();
           Block block1 = Block.Load("E:\\1.dxf", "1");
            Vector2 center1 = new Vector2(0, 0);
            Insert insert1 = new Insert(block1, center1);

           Block block2 =Block.Load("E:\\2.dxf", "2");
            Vector2 center2 = new Vector2(0, 500);
            Insert insert2 = new Insert(block2, center2);

           dxf.Save("E:\\3.dxf");

Results:
[image]

related dxf files
[dxf.zip](https://github.com/haplokuon/netDxf/files/7445574/dxf.zip)

haplokuon, 2021-10-31:
Read [this post](https://github.com/haplokuon/netDxf/issues/324), you have the same problem. Both files 1.dxf and 2.dxf have a block definition named "*U1" that comes from a dynamic block, when you try to merge them into the same document one overwrites the other. You need to rename them first, this library does not support dynamic blocks.

seppelhuang, 2021-10-31:
How to rename the blocks using this library, Using Block.Name="xxx" throw exception " "Blocks for internal use cannot be renamed."

haplokuon, 2021-10-31:
You need to use the latest source code, check the changelog.txt

* The internal blocks that start with "*U" or "*T" can be safely renamed. They are internally created to represent dynamic blocks, arrays, and tables; although the information of those objects is lost when importing the DXF, the block that represent its graphical appearance is imported.