haplokuon/netDxf #329: conditional compilation #if !NET45 should be enhanced

LeiYangGH, 2021-11-03:
If install vs2019 on a new Win10, .net 4.5 won't be installed by default(and even cannot be installed). But .net 4.8 can be installed.

> netDxf\netDxf\IO\DxfReader.cs

```cshar
#if !NET45
                Encoding.RegisterProvider(CodePagesEncodingProvider.Instance);
#endif
```
The code won't compile if re-target the project to .net 4.5.2, 4.6.1, 4.8 etc.

haplokuon, 2021-11-08:
If you need to target a different framework you will need to manually edit the "netdxf.csproj" file.

LeiYangGH, 2021-11-09:
yes. but after editing it, the code won't compile. The best compability is you  just need change the csproj, without other code. This is just enhancement, though.

haplokuon, 2021-11-10:
The only thing you need to do is to change the target framework for 4.5 'net45' for the framework  4.8 'net48' in both places of the netdxf.csproj. There is no need to rename the constant 'NET45', that is to handle the changes between the net framework 4 family and previous ones; and the net Standard, net Core, and net 5, and I guess later versions like net 6.

LeiYangGH, 2021-11-10:
I have tried so i'm afraid not like that.
`Encoding.RegisterProvider` should be compiled under .net core, right?
In .net 4.8, for example, it cannot compile.

LeiYangGH, 2021-11-11:
ok. i missed another NET45 in the csproj. So `NET45` here means 'non-standard', not the target framework itself, right? I think the problem is solved. Thanks!

closed 2021-11-11 by LeiYangGH