haplokuon/netDxf #330: Problems accessing SPLines FitPoints

laihuaxun2004, 2021-11-06:
Hello, I am a beginner in China. I found this problem when using this netdxf library:
When some graphics are spline curves generated by other software, no fitting points are generated in the dxf file, you need to open it with the higher version of AutoCAD software, change the display mode of the spline curve from control points to fitting points, and then save It is a DXF file. At this time, there will be fitting point data in the dxf file, but the netdxf library cannot read these fitting points.
I don't know if it is a problem with my program, or a problem with nextdxf. This passage was translated through Google translation software, I don’t know if it is correct. Thanks!

haplokuon, 2021-11-08:
I do not see any problems reading the fit points, just make sure they are actually present in the DXF.

laihuaxun2004, 2021-11-12:
I may be mistaken, there is no fitting point of the spline curve in the DXF file generated by the higher version of autoCAD

closed 2021-11-12 by laihuaxun2004