haplokuon/netDxf #331: "Description" value of layer

pmorshed, 2021-11-11:
Hi. How can i get "description" property for a layer?! 
netDxf.Tables.Layer class contains the "Name" property but doesn't contain the "Description"...

haplokuon, 2021-11-14:
In the official DXF documentation there is no code for the layer description, but after checking if that value was really stored in the DXF, I found that it really does but in the extended data information. Another example of Autodesk at its best, they had to choose the worst way of saving a simple string. Why not do the same as the line type or the block which descriptions are saved under a code 3 and 4 respectively?

At the moment if you need the description you will have to read it manually from the XData.

pmorshed, 2021-11-14:
Thanks
How can i read it from XData? Do you have an example?

haplokuon, 2021-11-16:
```c#
DxfDocument doc = DxfDocument.Load("Drawing1.dxf");
Layer layer = doc.Layers["Layer1"];
// Get the extended data information,
// how and where the layer description is stored is undocumented, proceed with care
// the layer description is saved as extended data information associated to an ApplicationRegistry called "AcAecLayerStandard"
string layerDescription;
XData xData = layer.XData["AcAecLayerStandard"];
foreach (XDataRecord data in xData.XDataRecord)
{
	// looks like it just stores two strings values under a code 1000
	// the first one seems to be always empty, I do not know if it is there just to waste space
	// or it is actually used for something else from who knows where
	if (data.Code == XDataCode.String)
	{
		layerDescription = data.Value as string;
	}
}
```
If you need more detailed information about the extended data go to the AutoCad manual, you can start [here](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-A2A628B0-3699-4740-A215-C560E7242F63).

pmorshed, 2021-11-16:
Many thanks. You’re awesome