haplokuon/netDxf #334: Vertices coordinates in UCS (not WCS)

pmorshed, 2021-11-11:
I have a dxf file that WCS and UCS are not in the same position/direction. 
So, how can in get vertices coordinates of entity in UCS (user coordinate system) of DXF file?

haplokuon, 2021-11-14:
If you need to convert WCS coordinates to UCS you will have to do it manually. It has been a long time since I look into the UCS table, and I may add, in a future update, some kind of Transform method for the UCS object to convert coordinates.

pmorshed, 2021-11-14:
But US table needs "name" for finding a ucs. what about "unnamed" ucs?!

haplokuon, 2021-11-16:
If you are looking for the current UCS (the last UCS when the drawing was saved) the basic information is stored in the DrawingVariables.
```c#
DxfDocument doc = DxfDocument.Load("Drawing1.dxf");
Vector3 ucsOrigen = doc.DrawingVariables.UcsOrg;
Vector3 ucsXDir = doc.DrawingVariables.UcsXDir;
Vector3 ucsYDir = doc.DrawingVariables.UcsYDir;
```
With that information you will have to manually build your transformation matrix to convert your coordinates from WCS to UCS or vice versa.

pmorshed, 2021-11-16:
Thanks so much