haplokuon/netDxf #335: change assertions(that depend on user inputs) to exceptions if possible

LeiYangGH, 2021-11-21:
I got some dxf files that has some issues, but can open with cad viewer. when open with netdxf assertion error prompt up and entire program cannot continue.
I have many good files and some bad files, to find the bad ones, i use some code like this:
```csharp
                foreach (string dxfFile in Directory.GetFiles(dir, "*.dxf", SearchOption.AllDirectories))
                {
                    try
                    {
                        DxfDocument doc = DxfDocument.Load(dxfFile);
                    }
                    catch (Exception ex)
                    {
                        Console.WriteLine(dxfFile );
                        Console.WriteLine(ex.Message);
                    }
                }
```
but assertion makes it impossible.
[image]

Could the assertion be replaced with exceptions(return null DxfDocument is acceptable)?
The bad file sample is attached
[assertfail.zip](https://github.com/haplokuon/netDxf/files/7576484/assertfail.zip)
.

haplokuon, 2021-11-24:
You should use the Release build of the library, the Debug build is not intended to be use by you, unless you want to get your hands dirty, it is for testing purposes only. I use assertions through the DXFReader to notify of bad data values in the file that were given a default value, and I want to keep track of them in case it has some side effect down the line. In your case your file contains multiple Style objects that has neither a font file nor font family information, so the default "simplex.shx" font file is used instead.

LeiYangGH, 2021-11-24:
ok. thank you very much!

closed 2021-11-24 by LeiYangGH