haplokuon/netDxf #338: After saving DXF, some objects' images are dostorted.

DikBSD, 2021-11-30:
After saving DXF, some objects' images are dostorted.
**OS**: Windows 10 x64
**Net DXF version**: 3.0.0.0
Code:

```
using netDxf.Entities;
using netDxf.Header;
using netDxf.Objects;
using netDxf.Tables;
using netDxf.Units;
using Attribute = netDxf.Entities.Attribute;
using FontStyle = netDxf.Tables.FontStyle;
using Image = netDxf.Entities.Image;
using Point = netDxf.Entities.Point;
using Trace = netDxf.Entities.Trace;

namespace netDXFOnlyLoadSaveFile
{
    class Program
    {
        static void Main(string[] args)
        {
            DxfDocument dxf = DxfDocument.Load("in.dxf");
            dxf.Save("out.dxf");
        }
    }
}
```

1. We load in.dxf.
2. Save it immediately into out.dxf.
3. Open both in.dxf and out.dxf with AutoCAD.
4. Compare image of the objects in both files.

Screenshot of obects' images before saving.
[image]

Screenshot of obects' images after saving.
[image]

in.dxf and out.dxf for problem testing is attached

[in.dxf.zip](https://github.com/haplokuon/netDxf/files/7623389/in.dxf.zip)
[out.dxf.zip](https://github.com/haplokuon/netDxf/files/7623390/out.dxf.zip)
.

haplokuon, 2021-12-02:
When working with shapes either as a Shape entity or as part of a complex line type, the .SHX file needs to be accessible. In your case, you are using a shape in complex line types that depends on the ltype.shx file. You can pass a list of folders to the DxfDocument construtor where to look for that file. Something like:
```c#
DxfDocument doc = DxfDocument.Load("in.dxf", new []{@".\support"});
```

KierenTinning, 2021-12-07:
Did this work for you? We had the same problem and we included the ltype.shx file but it didn't appear to function