haplokuon/netDxf #347: Line with Text not outputting

KierenTinning, 2022-01-07:
This is related to 

https://github.com/haplokuon/netDxf/issues/338

In my scenario, the DXF does not output the line with text data - which is present in the original DXF fil
[Fonts (2).zip](https://github.com/haplokuon/netDxf/files/7829126/Fonts.2.zip)

The attached showss the original file, but when we export to DXF

[dataengine-2021-12-17-12-53-47.zip](https://github.com/haplokuon/netDxf/files/7829129/dataengine-2021-12-17-12-53-47.zip)

See attached, the Line with text is eliminated and it simply shows the line - we've included the additional font and linetype files in the system as per issue 338 but it does not work.

Is this function not supported? Or are we doing something incorrect.

Thank you

haplokuon, 2022-01-12:
I am unable to reproduce your problem. The DXF file in the "dataengine-2021-12-17-12-53-47.zip" somehow lost the text information of the complex linetype segments. I had to create a new DXF from the DWG included in the "Fonts (2).zip" and that worked without problems. Make sure you are using the latest source code.

The https://github.com/haplokuon/netDxf/issues/338 issue is not related with complex linetypes that contains a text string, only when using shapes as a Shape entity or as a shape in a complex linetype.

KierenTinning, 2022-01-12:
Correct, if we start with a Dxf that has the text in the line, the netDXF solution appears to drop that information.  Do you have an example that works?  We are working with the latest, but certainly seem to be doing something incorrectly.

haplokuon, 2022-01-12:
I do not understand. If you have a problem saving a DXF with netDxf from a previously loaded file, post the original DXF and I will take a look. If you have problems with a manually created one, post the source code.

KierenTinning, 2022-01-12:
[Assessment.zip](https://github.com/haplokuon/netDxf/files/7856313/Assessment.zip)
Attached is the source file (DXF) loaded into the system, it has the text inside the lines - when we save it out (the dataengine file) you can see the text gets dropped from the lines

KierenTinning, 2022-01-12:
When we include the data from the fonts file, the text still gets dropped - which is where we think we must be missing something

KierenTinning, 2022-01-18:
So, in your project, you can load this file and output it?

KierenTinning, 2022-02-08:
Figured it out - turns out the nuget package was out of date, if we use the sample code it works.  

Can I ask if .net 4.7.2 is supported?  I know officially not, but will it work?

haplokuon, 2022-02-13:
So, when I ask you if you were using the latest source code you were not. I am not related to any of the nuget packages that has been published, I only upload here at GitHub.

Read [this](https://github.com/haplokuon/netDxf/issues/329). Instead of substituting the target framework 'net45' by 'net48' like in the referenced post use 'net472' to target the Net Framework 4.7.2. The library should work just fine for any version of the Framework 4 family. Although, the project is multitarget, I will not write a target for every single net framework version around.

closed 2022-02-14 by KierenTinning