haplokuon/netDxf #357: without version is it possible to open the dxf file

Vaibhav-Goti, 2022-02-14:
Hello

i try to open one dxf file but in this i facing some issue like version is unknown in exception so i want to open this as default version what is procedure for this please help for this.

haplokuon, 2022-02-19:
No, and it is a little more complex that just the version. If your file is a valid DXF, I guess you are dealing with a pre-2000 version.

Vaibhav-Goti, 2022-02-22:
yes. ok so how can i open this file.
i am attaching hear file please try to open and what is solution for this.
[sample.zip](https://github.com/haplokuon/netDxf/files/8114254/sample.zip)
 please help for open it.

haplokuon, 2022-02-27:
You will need to convert your file to an AutoCad 2000 version or newer. The file you posted only contains polylines that you can easily convert them with QCad or LibreCAD if you need a free alternative.

Your file actually contains a header with the version information but so old that it does not appear in the AutoCad DXF documentation. Checking in the Wikipedia the version of your file (AC1003) corresponds to AutoCad 2.6 released in 1987.