haplokuon/netDxf #365: Latest version on Nuget

ktheijs, 2022-03-09:
I am using netDxf through Nuget and I see that it is falling behind quite a bit. I have an issue where I do arcs/splines/... ToPolyline() but it returns a wrong result when the Normal of that entity is not (0,0,1). I believe this issue has been fixed in one of the latest commits.

Would it be possible to create a new release that includes all the latest commits and make it available on Nuget?

GinoDerycke, 2022-03-11:
Yes, I was wondering too how to get the latest version on Nuget. I like the package but I would prefer to be able to download the latest stable version from Nuget.

haplokuon, 2022-03-11:
I do not publish anything outside Github. The packages at Nuget are not mine, you will have to contact its actual owner.

ktheijs, 2022-03-11:
Do you know who that is? On Nuget is says that you 'Daniel Carvajal' is the owner.

haplokuon, 2022-03-11:
As far as I know there are three at nuget and none of them says that I am the owner. Go to the nuget website and search for them.

JackClouseau, 2022-04-01:
I think 'thieum' is the owner of netDXF.
https://www.nuget.org/profiles/thieum

droid1700, 2022-12-21:
I contacted thieum. A new version was posted today

GinoDerycke, 2022-12-22:
Great, thx! Will the updates now be done periodically again on Nuget?

dojo90, 2023-04-19:
> As far as I know there are three at nuget and none of them says that I am the owner. Go to the nuget website and search for them.

Why don't you publish it on nuget?

DavidWishengrad, 2023-04-19:
Hey Dojo,

That question is for those who are putting his code elsewhere.  He wrote
the code, has others that contribute here, publishes here and treats people
well. That's more than enough.

Please consider using the search here. In your case with the word 'nuget'.

Also, has just recently noted that he is very busy.

Thank you for understanding. 👍

dojo90, 2023-04-19:
@DavidWishengrad 
Well, it is of course very complicated to integrate a "nuget deploy job" into the build pipeline. *irony off*

But of course everyone has a different opinion on how to publish and present their library.

Thanks for your feedback anyway.

haplokuon, 2023-04-19:
@dojo90 
Well, it is of course very complicated to download the netDxf project and press Build, _irony off_

If you are so interested in nuget why don't you do it? It is not like the MIT license does not allow it.