haplokuon/netDxf #369: Special file(attached) can not be open(cad not responding) after read and write with netdxf

LeiYangGH, 2022-03-16:
Hi I've been using your library for about one year, and works quite well, thanks!
Recently got some file, which can be opened by cad viewer(2022), but when read and written by netdfx, cad viewer cannot open it any more(hang forever when opening).
I know there should be some issue in the original file, but I have no clue. 
Could you have a check?

```csharp
                    DxfDocument doc = DxfDocument.Load(dxfFile);
                    doc.Save(dxfFile);
```

LeiYangGH, 2022-03-16:
[test.zip](https://github.com/haplokuon/netDxf/files/8262253/test.zip)

haplokuon, 2022-03-31:
Try the latest source code, if you haven't done so it is the first thing you should try. The only thing I see is that your DXF file contains a Polyline2D with 0 vertexes. With the latest code trying to save your file showing the handle of that problematic entity that you will have to manually remove. In the next update such entities will be automatically removed when saving only throwing an assertion under Debug mode. Besides this, I do not see anything else, your problem might be due to an issue that has already been fixed.

LeiYangGH, 2022-04-01:
thanks! I was always using latest code when I create this question. I also tried latest tagged dll, but its interface is incompatible with latest code(i remember `.AddEntityObject` and `EntitieObjects.Add` difference, and my work heavily depend on your latest code's rename EntityObject feature, so I confirm I'm using latest code.

I'll try to find that problematic Polyline2D by code and remove, thanks!

LeiYangGH, 2022-04-07:
seems removing the problematic object solves the problem, thanks!