haplokuon/netDxf #388: Block Name not found

Chiron67, 2022-06-24:
Hello please can you help me.
i got this exception 
System.Exception : 'The block record UNFOLD is not defined.'

[W-07-027 S.3 POS 40.dxf.zip](https://github.com/haplokuon/netDxf/files/8974441/W-07-027.S.3.POS.40.dxf.zip)
just remove the .zip extension .dxf upload are not allowed

thank you

haplokuon, 2022-06-26:
Your file is pretty broken, it lacks the BLOCK_RECORD TABLE for the UNFOLD block. You can try it with Autodesk DWG TrueView it will show the same issue and will discard the drawing.

Chiron67, 2022-06-27:
Hi ,
thanks for your reply and anotation,
i have installed the autodesk DWG Trueview, but i cannot follow up your lack problem, the drawing is still visible.
[image]

using edrawing and also autocad do not give me some Block_record anomaly.
[image]

[4425-Test-S-150x150.zip](https://github.com/haplokuon/netDxf/files/8989735/4425-Test-S-150x150.zip)

maybe the file was corrupted when uploading , i have zipped them
thank you

haplokuon, 2022-06-28:
This second file you posted is new and different from the one in your first post. In this case the file has been saved as a pre-AutoCad2000 version. This library is only compatible with versions AutoCad2000 and upwards.