haplokuon/netDxf #389: Missing Attribute definations

Nikhilsp, 2022-07-05:
A few of the spec texts are missing in my dxf. After debugging I found that there is no attribute in dxf.Entities.AttributeDefinitions which has the handle value as same as of those text. I have attached the part of dxf below. Text "NIC2324" is not getting displayed on screen. What could be the possible reason for missing attribute definitions or where should I look?

 0
ATTDEF
  5
EBA7
330
2898
100
AcDbEntity
  8
0
  6
Continuous
 62
     7
370
    -3
100
AcDbText
 10
257.94316280237
 20
484.7253392768378
 30
0.0
 40
2.5
  1

 50
90.0
 41
0.8
 72
     2
 11
258.7764961357033
 21
491.3920059435044
 31
0.0
100
AcDbAttributeDefinition
  3
ENTER SPEC TAG 1
  2
NIC2324
 70
     0
 74
     1
280
     0
 71
     1
 72
     0
 11
258.7764961357033
 21
491.3920059435044
 31
0.0

sebastiankozub, 2022-07-06:
I noticed that when there is few attribute definitions containing same:
3 | Prompt string and/or 2 | Tag string (cannot contain spaces)
the DxfDocument contain only one of those attributes...

sebastiankozub, 2022-07-06:
Having those 2 attribute definitions in DXF file. DxfDocument contain only one of them:

First: 

> ATTDEF
>   5
> 114DE
> 330
> 5308
> 100
> AcDbEntity
>   8
> 0
>   6
> Continuous
>  62
>      7
> 370
>     -3
> 100
> AcDbText
>  10
> 194.3718116027041
>  20
> 201.6237524133556
>  30
> 0.0
>  40
> 2.5
>   1
> 
>  50
> 90.0
>  41
> 0.8
>  72
>      2
>  11
> 195.2051449360374
>  21
> 206.2904190800223
>  31
> 0.0
> 100
> AcDbAttributeDefinition
>   3
> ENTER SPEC TAG 1
>   2
> 1G4
>  70
>      0
>  74
>      1
> 280
>      0
>  71
>      1
>  72
>      0
>  11
> 195.2051449360374
>  21
> 206.2904190800223
>  31
> 0.0
>   0
> ATTDEF

Second:

> 
> ATTDEF
>   5
> 1147A
> 330
> 5308
> 100
> AcDbEntity
>   8
> 0
>   6
> Continuous
>  62
>      7
> 370
>     -3
> 100
> AcDbText
>  10
> 423.9105809632231
>  20
> 150.4010964133757
>  30
> 0.0
>  40
> 2.5
>   1
> 
>  50
> 90.0
>  41
> 0.8
>  72
>      2
>  11
> 424.7439142965565
>  21
> 155.0677630800424
>  31
> 0.0
> 100
> AcDbAttributeDefinition
>   3
> ENTER SPEC TAG 1
>   2
> 1G4
>  70
>      0
>  74
>      1
> 280
>      0
>  71
>      1
>  72
>      0
>  11
> 424.7439142965565
>  21
> 155.0677630800424
>  31
> 0.0
>   0
> ATTDEF

Any hints for a code file or line that needs to be debugged or changed to have both definitions in the DxfDocument collections?

haplokuon, 2022-07-07:
Attribute definitions with the same tag are not allowed and will be automatically deleted, this is on purpose. I know that the DXF allows it but I don't. The main problem with it is that the tag string is the only way to reference the attribute to the original attribute definition, therefore if multiple attribute definitions can have the same tag that link is lost. If you modify any attribute definition of the block there will be no way to propagate the changes to all inserts that references the block. You still have access to all attributes of the insert, regardless if they were created by an attribute definition with a duplicate tag or not.

Sometime, in the past, the AutoCad developers realize that it was not a good idea and it will show in red in the block attribute manager attribute definitions with the same tag. This is something you can read in the AutoCad official documentation "Identify duplicate tag names. Because duplicate tag names can lead to unpredictable results, you can set the Block Attribute Manager to highlight duplicate tags so that you can change them." 

So, unless, there is a very good reason I will keep it the way it is. It is a very bad practice duplicating tags.

@Nikhilsp In general a piece of a DXF has very little use, post the whole DXF instead.

sebastiankozub, 2022-07-08:
Thanks @haplokuon for quick and comprehensive answer that gives clues to make decisions - generally we need those entities so probably forking your solution and parametrize the constructor :)