haplokuon/netDxf #391: DXF file handling using C#.

rspardeshi, 2022-07-08:
hi,
I am trying to read the dxf file in C# and want to read its universal coordinate system.
please give me suggestions....

thanks.

haplokuon, 2022-07-10:
You will need to be a lot more specific with your questions. As a general rule, if you see a Vector2 as a property of an entity it is defined in local/object coordinate system OCS, if you see a Vector3 it is in world coordinate system WCS.

rspardeshi, 2022-07-12:
Thanks for the advice.....actually i have developed one tool which is helpful for arranging the multiple dxf file into one single dxf file. But the problem is if any dxf file UCS is not set correctly or misplaced then that drawing will not be getting placed next to its respective dxf drawing, it will be placed elsewhere. So I want to read the dxf files and relocate their UCS or WCS coordinates at the required position so that all dxf files will be getting arrange in a proper manner.

For example-1 with all dxf having same WCS or UCS coordinates equal gap between each dxf file vertically and horizontally:- 
Note - this output was achieved by me...

dxf6   dxf7  dxf8  dxf9  dxf10

dxf1   dxf2  dxf3  dxf4  dxf5

For example-2 with one or two dxf having different WCS or UCS coordinates, here dxf2, dxf5 and dxf10 having different UCS than other dxf files:- 
Note - trying to solve this issue...

dxf6  dxf7  dxf8  dxf9            dxf10
    dxf2 dxf5
dxf1   dxf3  dxf4  

thanks...

haplokuon, 2022-07-13:
Taking into account what I said in my previous post, in a DXF, all points and vectors are saved in the same coordinate system (WCS). Therefore, if a line has a start point of (1,1,1) in WCS, it will always be (1,1,1) no matter the DXF.

A drawing may contain multiple user coordinate system (UCS) that you can find in the UCSs collecition of the drawing; additionally, the drawing can have an active UCS that you can find in the drawing variables property CurrentUCS. So, the transformations between the WCS and UCSs must be performed manually when needed. The class UCS contain a couple methods to help you with this task.

rspardeshi, 2022-07-25:
thanks UCS and WCS related issue solved.

Now I have some dxf files which is generated by Tekla Structure Software version 2021.
For this dxf files i am not able read there "$EXTMIN" and "$EXTMAX" values since values are not in proper format.

example :- 
                 
$EXTMIN
 10
1.000000000000000E+20
 20
1.000000000000000E+20
 30
1.000000000000000E+20
  9
$EXTMAX
 10
-1.000000000000000E+20
 20
-1.000000000000000E+20
 30
-1.000000000000000E+20
  9

But when I was open this file in AutoCAD and saved it after that I am able to read its  $EXTMIN and $EXTMAX values.
So could you please provide me any solution for above issue.
Is it possible to open the dxf file and save it and again use this file for further task without losing its data.

Thanks.