haplokuon/netDxf #393: not able to read "$EXTMIN" and "$EXTMAX" values from DXF files.

rspardeshi, 2022-07-25:
Hi,
Now I have some dxf files which are generated by Tekla Structure Software version 2021.
For these dxf files, I am not able to read their "$EXTMIN" and "$EXTMAX" values since values are not in the proper format.

example :-

$EXTMIN
10
1.000000000000000E+20
20
1.000000000000000E+20
30
1.000000000000000E+20
9
$EXTMAX
10
-1.000000000000000E+20
20
-1.000000000000000E+20
30
-1.000000000000000E+20
9

But when I opened this file in AutoCAD and saved it after that I am able to read its $EXTMIN and $EXTMAX values.
So could you please provide me with a solution for the above issue?
Is it possible to open the dxf file and save it and again use this file for another task without losing its data?

Thanks.

rspardeshi, 2022-07-27:
First thanks for the response.

My problem is when my tool trying to read dxf file which is exported from Tekla Structure that time its EXTMIN and EXTMAX values are not in the proper format they are like "1.000000000000000E+20" this kind of number, but when I open the same dxf file in Autocad that time these 2 variables having proper decimal values and also readable through my tool.
Following are those 2 variables and their values.

$EXTMIN
 10
3392.128751969677
 20
4474.958629929921
 30
0.0
  9
$EXTMAX
 10
4475.846394006834
 20
4788.772733162242
 30
0.0
  9

so I am asking you for advice if you have any ideas or suggestions, instead of manually opening and saving those dxf files which are exported from Tekla Structure how can I automate this task using netDXF or any other way.

thanks.

haplokuon, 2022-07-29:
Why the scientific notation is causing you problems? Writing a million as 1000000 or 1e6 are both valid ways of expressing numbers, so whatever you are doing you should be able to handle them. All programming languages will manipulate them easily, at least the ones I know of.

rspardeshi, 2022-08-01:
Thanks for the response I have solved the issue now.