haplokuon/netDxf #402: Ask Help For Text/Mtext total length.

airsxue, 2022-08-14:
Hello,
Thanks for your hard work for netDxf,I just want to know how can I get the **total Length** of a text or Mtext(as below pic. )? Thanks!
[image]

airsxue, 2022-08-14:
> Hello, Thanks for your hard work for netDxf,I just want to know how can I get the **total Length** of a text or Mtext(as below pic. )? Thanks! [image]

BTW, I used tried to get the total Length by the "total Char count"* "width factor", but the result sstill has deviation with actual length.

airsxue, 2022-08-14:
> Hello, Thanks for your hard work for netDxf,I just want to know how can I get the **total Length** of a text or Mtext(as below pic. )? Thanks! [image]

sorry, correct the method i used tried, it should be **"total Char count"* "width factor" * textHeight**

Snailya, 2022-08-24:
As far as I know, the length of the text is depends on the system's graphic render.
So there's no method to get the text length before render.
In windows, you can do this by gdi or gidplus.  see https://stackoverflow.com/questions/263614/calculate-the-display-width-of-a-string-in-c

airsxue, 2022-08-27:
> As far as I know, the length of the text is depends on the system's graphic render.
> So there's no method to get the text length before render.
> In windows, you can do this by gdi or gidplus. see https://stackoverflow.com/questions/263614/calculate-the-display-width-of-a-string-in-c

Got it with many thanks

haplokuon, 2022-08-27:
Both SHX and TTF fonts stores vector data information from which you can, from every single character (glyph), build words, sentences, paragraphs,... and get its measurements; but you will need to use other libraries for that. So it is not exactly true that the length of the text depended on the system graphic render, although its is common in them to find some kind getLength method.

haplokuon, 2022-08-27:
Both SHX and TTF fonts stores vector data information from which you can, from every single character (glyph), build words, sentences, paragraphs,... and get its measurements; but you will need to use other libraries for that. So it is not true that the length of the text depends on the system graphic render, although its is common in them to find some kind getLength method.

airsxue, 2022-09-15:
> Both SHX and TTF fonts stores vector data information from which you can, from every single character (glyph), build words, sentences, paragraphs,... and get its measurements; but you will need to use other libraries for that. So, it is not true that the length of the text depends on the system graphic render, although it is common in them to find some kind getLength method.

so may i have your suggestion on  which libraires can i use to measure the length of text?  thanks