haplokuon/netDxf #407: DXF file modification/editing.

rspardeshi, 2022-08-22:
Hi Haplokuon,

Is it possible to update, edit, or alter the DXF file data using netDxf ?
I want to select a particular Polyline and wish to delete that polyline from the selected DXF file.

thanks,
Rahul Pardeshi.

haplokuon, 2022-08-22:
Yes, the the purpose of this library is being able to create, modify, and reading DXF files. You just need to load your file, remove the entity, and save it again.

rspardeshi, 2022-08-25:
Hi Haplokuon,

thanks,
I am studying netDXF library and successfully created one tool using it.
Now I want to select a particular Polyline and wish to delete that polyline from the selected DXF file.
Could you please provide me any example or reference matching my task?

haplokuon, 2022-08-26:
I cannot tell you how to find a particular entity in the drawing you must know how to find it. You will have to decide if it has a particular color, if it is inside a unique layer, if it is by its geometry,...; although the most efficient way is by its handle.
```c#
DxfObject o = doc.GetObjectByHandle("");
doc.Entities.Remove(o as EntityObject);
```

rspardeshi, 2022-08-29:
thanks for the advice... it's constructive for my development.

my88480, 2023-04-20:
List<Polylines2D> allOuterContourEnts = DxfDoc.Entities.Polylines2D.Where(e => e.Layer.Name.Equals("0")).ToList();
DxfDoc.Entities.Remove(allOuterContourEnts);