haplokuon/netDxf #409: Show Dimentions

LauraBolanos05, 2022-08-24:
Hi there
I had a list with the Coordenates to create a square but I want to show the dimension of the lines, Using this code I can show the dimensions
**doc.AddEntity(entity);
doc.AddEntity(ad);**

But when I put everything together into a Block the dimensions does not show... 
//block.Entities.Add(entity);
//block.Entities.Add(ad);
//Insert insert = new Insert(block);
//doc.AddEntity(insert);

My final goal with this is to show the dimensions of the line with a specific width, in this moment the numbers look very small, I tried to change the height and width of the text using the AlignedDimension methods but anything had changed

Hope you can help me
Regards

DxfDocument doc = new DxfDocument();
foreach (List<Coordenates> _Coordenates in lst_Coordenates)
            {
                for (int i = 0; i < (_Coordenates.Count() - 1); i++)
                {
                    Coordenates initial = _Coordenates[i];
                    Coordenates next = _Coordenates[i + 1];

                    if (i < (_Coordenates.Count() - 1))
                    {
                        Line entity = new Line(new Vector2(initial.x_coordenate, initial.y_coordenate), new Vector2(next.x_coordenate, next.y_coordenate));
                        AlignedDimension ad = new AlignedDimension(new Vector2(initial.x_coordenate, initial.y_coordenate), new Vector2(next.x_coordenate, next.y_coordenate), 50);

                        ad.Style.TextStyle = TextStyle.Default;
                        ad.Style.TextStyle.Height = 50.0;
                        ad.Style.TextStyle.WidthFactor = 10.0;

                        ad.Style.TextHeight = 50.0;

                        ##with this I can show the dimentions of all lines but the measurement looks very small
                        **doc.AddEntity(entity);
                        doc.AddEntity(ad);**

                        //Block block = new Block("DimensionBlock" + Counter);
                        //block.Record.Units = DrawingUnits.Inches;
                        //AttributeDefinition attDefINCH = new AttributeDefinition("MyAttributeINCH" + Counter);
                        //attDefINCH.Height = 10.0;
                        //block.Entities.Add(entity);
                        //block.Entities.Add(ad);
                        //Insert insert = new Insert(block);
                        //doc.AddEntity(insert);
                    }
                }
            }

Snailya, 2022-08-25:
I‘ve meet a similar promblem, which my dimension displays as normal, but the text height of the attribute is reset to 1.

haplokuon, 2022-08-26:
About why the dimensions does not appear when embedded into a block, read this #342, you can also read #98 about other issues with dimensions inside blocks.

About why the change to the text height of the dimension style is not applied, read #375, a similar thing is happening when changing properties of the text style of the dimension style.