haplokuon/netDxf #412: AlignedDimension text upside-down in Rhino3D

novosadkry, 2022-09-05:
Hello,

I'm having troubles with AlignedDimension text displaying upside-down in Rhino3d. I've tried setting TextRotation property and TextPlacement style overrides, but none of them have worked for me.

Here's the image from Rhino3D. The text is upside-down and should be below the dimension line.

[image]

What is interesting, is that when I try the same with AutoDesk without any StyleOverrides or manual rotations, the text is indeed correctly aligned.

[image]

Here's a snippet of the code. I'm achieving the rotation by swapping the two line points (correct me if there is a better way)
```
var dimStyle = DimensionStyle.Iso25;
dimStyle.DimLineColor = AciColor.Green;
dimStyle.TextColor = AciColor.Default;
dimStyle.DimScaleOverall = 10;

var layer = new Layer("Dimensions")
    { Color = AciColor.Green };

const double offset = 50;

var line = (isHorizontal ? flipY : flipX)
    ? new Line(a, b)
    : new Line(b, a);

var dim = new AlignedDimension(line, offset)
{
    Layer = layer,
    Style = dimStyle,
    TextRotation = 180, // Doesn't work
    StyleOverrides =
    {
        // Doesn't work either
        new DimensionStyleOverride(DimensionStyleOverrideType.FitTextMove, DimensionStyleFitTextMove.OverDimLineWithoutLeader),
        new DimensionStyleOverride(DimensionStyleOverrideType.TextVerticalPlacement, DimensionStyleTextVerticalPlacement.Below)
    }
};

doc.Entities.Add(dim);
```

Is this a bug inside Rhino3D, or am I missing some setting which could overcome this issue?

Thank you.

haplokuon, 2022-09-23:
Do not assume that all programs will be capable of recreating the dimension exactly as AutoCad does, using all dimension and dimension style parameters. I do not know about Rhinoceros is has been a long time since the last time I use it.

The dimension style parameter that controls the vertical positioning of the dimension text is
```c#
style.TextVerticalPlacement = DimensionStyleTextVerticalPlacement.Below;
```
You have this parameter as a dimension override, if it doesn't work just try to modify directly the dimension style, instead, and see what happens.

You can also try to set these dimension style parameters to false, since you are using the Iso25 and by default they are set to true:
```c#
style.TextInsideAlign = false;
style.TextOutsideAlign = false;
```
The text in your first image is upside down because you are rotating it 180 degrees (TextRotation parameter of the dimension).

novosadkry, 2022-09-26:
Unfortunately, even after setting these parameters as you described, I'm still getting an upside-down dimension in Rhino.
For now, I'll just stick with AutoDesk as it seems that Rhino is in the wrong here.

Still, I must thank you for answering so quickly to my question. Have a nice day!

closed 2022-09-26 by novosadkry