haplokuon/netDxf #413: When modifying the document viewport , the spline entity in the block is changed

seppelhuang, 2022-09-14:
var dxfDocument = DxfDocument.Load(@"E:\\1.dxf");
            dxfDocument.Viewport.ViewHeight = 100000;
            dxfDocument.Viewport.ViewCenter = new Vector2(0, 0);

            dxfDocument.Save("E:\\"2.dxf");

[image]

after modifying viewport 

[image]

[dxf.zip](https://github.com/haplokuon/netDxf/files/9563078/dxf.zip)

haplokuon, 2022-09-23:
You are using AutoCad.,don't you? You must regenerate your drawing, "regen" command.

haplokuon, 2022-09-23:
You are using AutoCad, aren't you? You must regenerate your drawing, "regen" command.