haplokuon/netDxf #414: question about <Assertion failed>（断言失败）

airsxue, 2022-09-29:
When I use netDxf to read this dxf file(sample.zip), it will show below error with **Assertion failed**, but it still work
 effective when I enter "Ingnore" ,could you please help to advice how can I avoid this situation? or the dxf format reason? Thanks.
[image]
[sample.zip](https://github.com/haplokuon/netDxf/files/9670009/sample.zip)

haplokuon, 2022-09-30:
Use the Release build, make sure that in the Build tab of the project properties the "Define Debug constant" is unchecked.

haplokuon, 2022-09-30:
Use the Release build, make sure that in the Build tab of the project properties the "Define Debug constant" is unchecked.

Also you can open the netDxf.csproj file and make sure that in the property group for the desired build, DEBUG does not appear in the DefineConstants group.